tawnkramer/sdsandbox #1: [Feature Request] ROS integration using rosbridge

bobjohnson2040, 2017-03-01:
It would be awesome if this included a ROS integration. I searched on Github and found a few other repositories that integrated ROS and Unity using rosbridge.

http://wiki.ros.org/rosbridge_suite

https://github.com/search?utf8=%E2%9C%93&q=ros+unity

tawnkramer, 2017-03-02:
That's a great idea. I don't really know the first thing about ROS. Would be happy to take a pull request if you have the means to implement this. 

I will try to look into this after April. Focusing on another project until then.

tawnkramer, 2017-06-27:
Can you give me a use case? Would you like to Unity to take input from a ROS controller? I'm not sure what you have in mind.

bobjohnson2040, 2017-07-26:
The ideal setup would have your keras/tensorflow model publishing steering commands on ROS. Here is an example: https://github.com/udacity/self-driving-car/tree/master/steering-models/steering-node 

Then you would also publish the Unity images/sensor readings over ROS for the NN to subscribe to.

closed 2019-02-19 by tawnkramer