tawnkramer/sdsandbox #3: Manual Car

desireend, 2017-06-26:
If I would like to modify this project, would it be possible to add a manual car?

tawnkramer, 2017-06-26:
That's a fine idea. I can add it to the interface to make it easier, but it is possible with the current project. I think the Thunderhill scene is setup this way by default. But all scene could be setup with a nicer interface to choose between manual and PID driven modes.
I will try to add this soon.

tawnkramer, 2017-06-27:
There is manual control now in the main scene. You can use either keyboard or joystick inputs. When you use joysticks, you may need to edit the joystick input preferences to get the correct settings for your device. In the main menu, check Edit | Project Settings | Input. Then look at the Horizontal and Vertical Inputs.

closed 2017-06-27 by tawnkramer