tawnkramer/sdsandbox #8: SocketIO is slow, faster with just asyncio

NumesSanguis, 2018-01-26:
I found your amazing project when I was looking for Unity 3D - Python interaction, and I got your project to work with my code base. However, the framerate was too low for my application. Therefore, I replaced FLASK with its asynchronous variant, aiohttp. Unfortunately, I didn't see much improvement.

Finally I created a whole different setup with just asyncio in Python and a simple TCP connection. It might not be as advanced as SocketIO, but it might give you or people here inspiration for a faster setup: https://github.com/NumesSanguis/Basic-Unity3D-Python-server

NumesSanguis, 2018-01-26:
Feel free to close this when you've seen it.

tawnkramer, 2018-01-26:
I started with a pure tcp sockets implementation and I found it was faster. But I had problems with dropped connections for some users. Thanks for the pointer on asyncio, I haven't tried that yet. How much better was your framerate?

tawnkramer, 2018-01-26:
I'd like to see a pull request if you are able to share your changes.

NumesSanguis, 2018-01-26:
I haven't done any extensive tests with it, but text from my terminal instantly dissapeared because of the many "message received" with my current code. With SocketIO I saw it moving up. I'm about to update the files _aiohttp (which also uses asyncio, but with SocketIO), which is a stripped down version of your code. Please check that repository again tomorrow.

tawnkramer, 2018-01-26:
Cool. Thanks!
Here's the older version based on asyncore. But looks like asyncio might be the future.
https://github.com/tawnkramer/sdsandbox/blob/6564e88a7b9920727c35d48eee28c2c0876c6d1d/src/predict_server.py

NumesSanguis, 2018-01-26:
I updated the repository to include my Unity 3D - Python SocketIO with aiohttp and asyncio version. The code is basically a stripped down version of your code, so I think it shouldn't hard to change from FLASK to aiohttp. I didn't test it, but I think it will be slightly faster.

However, without SocketIO (with aiohttp), just asyncio was about 80x faster. See the notes in the README.md about an unprofessional measurement.

NumesSanguis, 2018-02-07:
I realized the comparison might not completely be fair, because in my code sample the network code runs in a separate thread, which may be the reason of the speed-up.

Now that I've been sending high volumes of data, I noticed that the slowest part of my code in Unity was because of `Debug.Log()`. Removing all those statements, I could speed up my code with a factor 100 or more. It's seriously weird why this takes up so much resources in Unity.

NumesSanguis, 2018-03-20:
I currently switched to this person's approach, using ZeroMQ for network messaging: [https://github.com/valkjsaaa/Unity-ZeroMQ-Example](https://github.com/valkjsaaa/Unity-ZeroMQ-Example)

You can use [ZeroMQ](http://zeromq.org/) for simple network paradigms, but way more complex ones without performance loss are also possible. The best part is that they provide libraries for many languages, and it doesn't require a messaging server.

tawnkramer, 2018-03-20:
Thanks for the update. I've used ZeroMQ before with success. I've had some
troubles with buffering though, where a high rate of messages caused quite
high latency. I will take a look at your approach. Thanks for sharing.

tawnkramer, 2018-10-09:
the donkey branch is back to using straight tcp sockets now. works much better than websockets. I will try to find the time to merge into master. The donkey branch is far ahead and quite different. It will be difficult.

closed 2018-10-09 by tawnkramer