tawnkramer/sdsandbox #11: Instructions not up to date

araffin, 2018-11-01:
Hello,

It seems that both requirements and instructions are not up to date.
The current version requires tensorflow>=1.10.1, custom version of donkey gym (https://github.com/tawnkramer/donkey_gym) and custom version of donkey car (https://github.com/tawnkramer/donkey/tree/master/donkeycar)
Also, it seems that predict_server.py is not the good script anymore (but rather tcp_sim_server.py, which in my case fails due to "Expecting property name enclosed in double quotes: line 1 column 67 (char 66) failed to read json")

Could you update the README instructions and requirements.txt ? (I'm still struggling to make it work)

tawnkramer, 2018-11-01:
> First, thanks for this very nice project.

Why dont you try setup and instructions here:

https://github.com/tawnkramer/donkey_gym

araffin, 2018-11-01:
>Why dont you try setup and instructions here

Well, I have installed that donkey_gym version (using pip install -e .), donkeycar, I am talking about the install and run instructions of this repo (Setup and Demo part of the README)

araffin, 2018-11-01:
One important thing: I am also referring to the "donkey" branch which seems to be the dev branch (master branch was not update since september 2017...).

araffin, 2018-11-04:
@tawnkramer I think the problem comes from my Unity version, I am using `Unity-2018.2.7f1`, on which version was it tested ?

tawnkramer, 2018-12-06:
Yes, sorry for the stale instructions in the donkey branch. It's kind of my personal branch to support the donkey community. I'm trying now to encourage people to use the pre-built binaries in the donkey_gym because it's easier to me to support.

Would welcome a PR to update install, thanks.

I have tried the recent builds of Unity. They should work fine. 

donkey_gym does use json to serialize data and send to the python process. Is there's some problem with the float notation between the Unity's json generator and the python, then that would cause some trouble.

If you would like to do a PR for the floating point string notation, that would be cool.

Sorry for delay in response!
Thanks the awesome stable baselines.

tawnkramer, 2019-01-18:
thanks for PR to fix json floating notation and install requirements!

PR #14 (2019-01-18): Update requirements
Fixes #11
closed 2019-01-18 by tawnkramer