tawnkramer/sdsandbox #12: Crash when running headless

Bleyddyn, 2018-11-05:
I'm using the latest release from donkey_gym on Mac OS X 10.13. It runs fine unless I try to use the --headless option. Then I get a crash. The following is from unitylog.txt. Lots more repeats of the first few lines.

RenderTexture.Create failed: format unsupported - RGBA4 UNorm (9).
 
(Filename:  Line: 432)

RenderTexture.Create failed: format unsupported - RGBA4 UNorm (9).
 
(Filename:  Line: 432)

RenderTexture.Create failed: format unsupported - RGBA4 UNorm (9).
 
(Filename:  Line: 432)

RenderTexture.Create failed: format unsupported - RGBA4 UNorm (9).
 
(Filename:  Line: 432)

RenderTexture.Create failed: format unsupported - RGBA4 UNorm (9).
 
(Filename:  Line: 432)

RenderTexture.Create failed: format unsupported - RGBA4 UNorm (7).
 
(Filename:  Line: 432)

Receiving unhandled NULL exception
Obtained 36 stack frames.
#0  0x0000010a275f5d in GenericDynamicVBO::DrawChunkInternal(DynamicVBOChunkHandle const&, VertexDeclaration*, DrawBuffersRange const*, int, unsigned int, unsigned int)
#1  0x0000010a275163 in DynamicVBO::DrawChunk(DynamicVBOChunkHandle const&, VertexDeclaration*, DynamicVBO::DrawParams const*, int)
#2  0x0000010a410299 in RenderDynamicBatch(GfxDevice&, RenderNodeQueue const&, BatchInstanceData const*, unsigned long, unsigned long, unsigned long, VertexInputMasks, ShaderChannelMask, VertexChannelsLayout const&, GfxPrimitiveType)
#3  0x0000010a40f48e in RenderMultipleMeshes(RenderNodeQueue const&, RenderMultipleData const&, VertexInputMasks)
#4  0x0000010a197ace in BatchRenderer::RenderBatch(BatchInstanceData const*, unsigned long, VertexInputMasks) const
#5  0x0000010a19c45f in ForwardRenderLoopJob(GfxDeviceAsyncCommand::ArgScratch*, GfxDeviceAsyncCommand::Arg const*)
#6  0x0000010a43ac41 in GfxDevice::ExecuteAsync(int, void (*)(GfxDeviceAsyncCommand::ArgScratch*, GfxDeviceAsyncCommand::Arg const*), GfxDeviceAsyncCommand::ArgScratch**, GfxDeviceAsyncCommand::Arg const*, JobFence const&)
#7  0x0000010a19ac99 in ForwardShaderRenderLoop::StartRenderJobs(JobFence&, bool, bool, ShaderPassContext&)
#8  0x0000010a19c92e in ForwardShaderRenderLoop::PerformRendering(JobFence&, ActiveLight const*, ShadowJobData const&, bool, bool, bool, Camera::RenderFlag)
#9  0x0000010a19dfec in DoForwardShaderRenderLoop(RenderLoopContext const&, dynamic_array<RenderObjectData, 0ul>&, CullResults const&, bool, bool, bool, Camera::RenderFlag)
#10 0x0000010a191e6c in DoRenderLoop(RenderLoop&, RenderingPath, CullResults&, ShadowMapCache&, bool)
#11 0x0000010a10925d in Camera::DoRender(CullResults&, SharedRendererScene const&, Camera::RenderFlag, Camera::PerformRenderFunction*)
#12 0x0000010a10ab89 in Camera::CustomRender(CullResults&, ShaderPassContext&, CameraRenderingParams const*, Camera::RenderFlag, Camera::PerformRenderFunction*, void (*)())
#13 0x0000010a109cb7 in Camera::StandaloneCustomRender(Camera::RenderFlag, Shader*, core::basic_string<char, core::StringStorageDefault<char> > const&, Camera::PerformRenderFunction*)
#14 0x0000010a115187 in CameraScripting::Render(Camera*)
#15 0x000001191d0dbe in  (wrapper managed-to-native) UnityEngine.Camera:Render () + 0x6e (0x1191d0d50 0x1191d0dfe) [0x10d806cc0 - Unity Root Domain]
#16 0x0000010f40cd3a in mono_jit_runtime_invoke
#17 0x0000010f539946 in mono_runtime_invoke
#18 0x0000010a45d0b0 in ScriptingInvocation::Invoke(ScriptingExceptionPtr*, bool)
#19 0x0000010a448259 in MonoBehaviour::CallUpdateMethod(int)
#20 0x0000010a1c0c4d in void BaseBehaviourManager::CommonUpdate<BehaviourManager>()
#21 0x0000010a32f87e in ExecutePlayerLoop(NativePlayerLoopSystem*)
#22 0x0000010a32f8a7 in ExecutePlayerLoop(NativePlayerLoopSystem*)
#23 0x0000010a32faa6 in PlayerLoop()
#24 0x0000010a920dda in -[BatchModePlayerApplication UpdatePlayer]
#25 0x007fff3c652549 in __NSFireTimer
#26 0x007fff3a4cc564 in __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__
#27 0x007fff3a4cc1e4 in __CFRunLoopDoTimer
#28 0x007fff3a4cbcda in __CFRunLoopDoTimers
#29 0x007fff3a4c347b in __CFRunLoopRun
#30 0x007fff3a4c2867 in CFRunLoopRunSpecific
#31 0x007fff3c5dae76 in -[NSRunLoop(NSRunLoop) runMode:beforeDate:]
#32 0x007fff3c5dad4e in -[NSRunLoop(NSRunLoop) run]
#33 0x0000010a920f51 in -[BatchModePlayerApplication StartPlayerLoop]
#34 0x0000010a92f433 in PlayerMain(int, char const**)
#35 0x007fff62490015 in start

tawnkramer, 2018-12-06:
This occurs because it tries to use rendering services when none exist. Perhaps there's a soln for this. That's why I left it in there. In some environments you might be able to use a dummy XWindows environment to fool the renderer.

tawnkramer, 2019-02-19:
I wasn't able to make this work. But this looks like a way forward. Please let us know if anyone has a recipe to make this work.
https://towardsdatascience.com/how-to-run-unity-on-amazon-cloud-or-without-monitor-3c10ce022639

tawnkramer, 2019-03-17:
Unity rep reports possible fix for this in late 2019. Here’s hoping.

simeneide, 2019-08-31:
where did they say late 2019? :) Just ran into the same problem and did not get the towardsdatascience link to work.

tawnkramer, 2019-08-31:
Unofficially on Twitter

closed 2020-07-29 by tawnkramer