tawnkramer/sdsandbox #13: Neural network model not responding

JackWang1997, 2018-12-06:
Hi!
It seems that my neural network is not responding to unity. I can use pid driving but when i use nn driving the car is not moving at all. I am trying to use the highway.h5 btw.
My terminal looks like this: 

wangshidadeMacBook-Air:src StarJack$ python predict_server.py ../outputs/highway.h5
Using TensorFlow backend.
WARNING:tensorflow:From /Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/backend/tensorflow_backend.py:1205: calling reduce_prod (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
2018-12-06 03:58:51.752401: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
WARNING:tensorflow:From /Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/backend/tensorflow_backend.py:1290: calling reduce_mean (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
(24046) wsgi starting up on http://0.0.0.0:9090

I think there should be a line saying like got a new client right? It is not appearing so I'm wondering if i can get any help... thanks a lot!

JackWang1997, 2018-12-06:
also when i am trying to train the my own model, it gives me an error like this 

(py36) wangshidadeMacBook-Air:src StarJack$ python prepare_data.py --clean --validation
gathering images ../sdsim/log/*.*
(py36) wangshidadeMacBook-Air:src StarJack$ python train.py new.h5
Using TensorFlow backend.
working on model new.h5
WARNING:tensorflow:From /Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/backend/tensorflow_backend.py:1205: calling reduce_prod (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
WARNING:tensorflow:From /Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/backend/tensorflow_backend.py:1290: calling reduce_mean (from tensorflow.python.ops.math_ops) with keep_dims is deprecated and will be removed in a future version.
Instructions for updating:
keep_dims is deprecated, use keepdims instead
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
cropping2d_1 (Cropping2D)    (None, 60, 160, 3)        0         
_________________________________________________________________
lambda_1 (Lambda)            (None, 60, 160, 3)        0         
_________________________________________________________________
conv2d_1 (Conv2D)            (None, 30, 80, 24)        1824      
_________________________________________________________________
elu_1 (ELU)                  (None, 30, 80, 24)        0         
_________________________________________________________________
conv2d_2 (Conv2D)            (None, 15, 40, 32)        19232     
_________________________________________________________________
elu_2 (ELU)                  (None, 15, 40, 32)        0         
_________________________________________________________________
conv2d_3 (Conv2D)            (None, 8, 20, 64)         51264     
_________________________________________________________________
elu_3 (ELU)                  (None, 8, 20, 64)         0         
_________________________________________________________________
conv2d_4 (Conv2D)            (None, 4, 10, 64)         36928     
_________________________________________________________________
elu_4 (ELU)                  (None, 4, 10, 64)         0         
_________________________________________________________________
dropout_1 (Dropout)          (None, 4, 10, 64)         0         
_________________________________________________________________
conv2d_5 (Conv2D)            (None, 4, 10, 64)         36928     
_________________________________________________________________
flatten_1 (Flatten)          (None, 2560)              0         
_________________________________________________________________
dense_1 (Dense)              (None, 1000)              2561000   
_________________________________________________________________
dropout_2 (Dropout)          (None, 1000)              0         
_________________________________________________________________
elu_5 (ELU)                  (None, 1000)              0         
_________________________________________________________________
dense_2 (Dense)              (None, 128)               128128    
_________________________________________________________________
elu_6 (ELU)                  (None, 128)               0         
_________________________________________________________________
dense_3 (Dense)              (None, 1)                 129       
=================================================================
Total params: 2,835,433
Trainable params: 2,835,433
Non-trainable params: 0
_________________________________________________________________
(None, 60, 160, 3)
(None, 60, 160, 3)
(None, 30, 80, 24)
(None, 30, 80, 24)
(None, 15, 40, 32)
(None, 15, 40, 32)
(None, 8, 20, 64)
(None, 8, 20, 64)
(None, 4, 10, 64)
(None, 4, 10, 64)
(None, 4, 10, 64)
(None, 4, 10, 64)
(None, 2560)
(None, 1000)
(None, 1000)
(None, 1000)
(None, 128)
(None, 128)
(None, 1)
found 105 files
num train/val 79 26
steps_per_epoch 0 validation_steps 0
Traceback (most recent call last):
  File "train.py", line 250, in <module>
    go(args.model, epochs=args.epochs, limit=args.limit, inputs=args.inputs, aug_mult=args.aug_mult, aug_perc=args.aug_perc)
  File "train.py", line 225, in go
    callbacks=callbacks)
  File "/Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/legacy/interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "/Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/models.py", line 1121, in fit_generator
    initial_epoch=initial_epoch)
  File "/Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/legacy/interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "/Users/StarJack/anaconda2/envs/py36/lib/python3.6/site-packages/keras/engine/training.py", line 1936, in fit_generator
    raise ValueError('When using a generator for validation data, '
ValueError: When using a generator for validation data, you must specify a value for `validation_steps`.

i didn't change any of the script so I'm wondering whats wrong with my thing...

JackWang1997, 2018-12-06:
And it says this when i open the project at first:
API Update Required 

This project contains scripts and/or assemblies that use obsolete APIs.

If you choose 'Go Ahead', Unity will automatically upgrade any scripts/assemblies in the Assets folder found using the old APIs. You should make a backup before proceeding.

(You can always run the API Updater manually via the 'Assets/Run API Updater' menu command.)

So does this means my problem is caused by the differences of unity? I am using this latest version of unity now.

tawnkramer, 2018-12-06:
Next you have to start Unity and then press the "Use NN Steering" button. Then it will connect. And you should see something like:

(53917) wsgi starting up on http://0.0.0.0:9090
(53917) accepted ('127.0.0.1', 53768)
connect  5c588ebf5560443087bb4175a7afd50c
fps 52.58995534440852
fps 60.38024416381007

I just tried this with latest code and Unity. Yes, you will have to hit a button when first opening project to update the API. That's normal and usually works just fine.

tawnkramer, 2018-12-06:
You need more data. It's calculated that you have zero validation steps
because it's rounding down an integer. Try with > 300 images. More is
better.

JackWang1997, 2018-12-08:
Hmmmm yes i found out why i couldn't train my model but i don't think that is the reason why i can't use highway.h5 model... i was following your instructions but i still can't get reply from the neural network. is there any way that i can check where goes wrong in the process that unity connects neural network?

tawnkramer, 2019-01-17:
The best way is to run both the Unity simulator and python process in debuggers. You can also try adding prints to see if that gives you a clue.

tawnkramer, 2019-02-19:
please reopen if you are still having problems.

closed 2019-02-19 by tawnkramer