tawnkramer/sdsandbox #15: Multiple Donkey Cars and random seed

araffin, 2019-01-16:
Hello,

As I'm not really confident with neither with Unity nor C#, I would be interested in knowing how difficult it is to (and also where to start/look at):
- have multiple donkey cars in the simulation (and maybe then change the camera), so we can simulate a "race"
- be able to seed the track generation (notably through the tcp interface to use it with the gym wrapper) so we can have the same track from one run to another

Thanks in advance ;)

tawnkramer, 2019-01-17:
I’ve been working towards multiple cars running at once. There’s code
support now for instantiating cars and tying a socket to each. I can add an
api to ask for a new car and give it an external ip to connect to for
control.

 I’m not sure how this would work in the gym framework. Would you create a
gym instance per agent connecting. And a single sim process where they
interact. It’s possible.

Determinism and/or jumping to a save point would allow the Uber Go Explore
algorithm which seems quite promising. I’m not sure how well it will work
in this case. We would need to hold fixed the time intervals for control
input. And even then, I’m not sure it would be repeatable. But we can try.

araffin, 2019-01-17:
>I can add an api to ask for a new car and give it an external ip to connect to for control.

That would be nice, thanks =) Please tell me if you need help.

> I’m not sure how this would work in the gym framework. Would you create a gym instance per agent connecting. And a single sim process where they interact. It’s possible.

I don't know yet, my high-level idea was to be able to control two cars independently (and I don't know how to change the camera view so you can see the result). I will certainly have to tweak a bit the gym interface, but the idea is that I would like to use only for testing/demo purpose at first.

>Determinism and/or jumping to a save point would allow the Uber Go Explore algorithm which seems quite promising. I’m not sure how well it will work in this case. We would need to hold fixed the time intervals for control input. And even then, I’m not sure it would be repeatable. But we can try.

Well, I was thinking of something simpler than jumping to a given state. I just wanted to be able to generate the same track (e.g. passing a random seed for level 0) so I can do a fair comparison between agents.

tawnkramer, 2019-01-17:
Ok, yes, for track generation we can definitely make that deterministic, I
hope.

For more than one bot I would either need to do split screen which only
work with up to 4 bots, or move to a single higher camera that keeps them
in view or cuts around.  With only two, split screen should work, if render
fast enough.

Adding API to make a new one is probably less than a days work. Will try to
do soon.

araffin, 2019-01-17:
> With only two, split screen should work, if render fast enough.

The split screen is a good idea =)

>Adding API to make a new one is probably less than a days work. Will try to do soon.

Thanks, and again if I can help at some point, just ask ;)

tawnkramer, 2019-01-18:
There's some work checked into donkey branch now to support two cars. The new API is named "new_car" and takes a "host" and "port" string. If you send that to the sim, it will spawn a new car and have it connect to that host, and port and configure things for split screen. Right now the example use of this is in the 'predict_server.py' script with the optional --num_cars=2 argument.

Perhaps you can implement the needed changes to the donkey_gym code? Basically one client has to connect the server. Once connected it can ask for as many others as needed. And those others should be up and listening for connections before the cars are spawned. Hope that makes sense.

It would be cleaner if the Sim was the server and the model predictors were clients. It's a bit backwards now, but it works.

araffin, 2019-01-18:
Wow, thanks, that was fast.

>Perhaps you can implement the needed changes to the donkey_gym code? 

Sure, I'll first try in my repo this weekend and then create a PR for donkey_gym when I have something more or less clean.

tawnkramer, 2019-02-19:
multiple cars and support for random seed has been added.

closed 2019-02-19 by tawnkramer

araffin, 2019-02-19:
@tawnkramer yes, thanks for the support. I recently tried the seeding but was not successful :/ I will try to give you a better feedback this weekend ;)

tawnkramer, 2019-02-20:
There is a timing needed for this to work. The message needs to be sent
after the road level is loaded. Check the example code in my last commit.

araffin, 2019-06-09:
> I recently tried the seeding but was not successful :/ 

I finally found why it was not working (see PR #27 )  =)

tawnkramer, 2019-06-09:
Thanks for fixing!