tawnkramer/sdsandbox #18: How to control the MacOs simulator over network

jrabary, 2019-02-20:
Hi all,
I can't figure out how the control the simulator over network on the macOS version. 
As far as I understand, I have to run a predict server that will receive the observations from
the simultator through websocket. But, I'm not sure if the web socket client in the simulator is really
working and not easy to guess the port number it uses.
Any idea on that ?

tawnkramer, 2019-02-28:
you might want to try https://github.com/tawnkramer/donkey_gym. It's a gym wrapper around this project, but it also uses raw tcp and works a bit better.

tawnkramer, 2019-03-01:
I've just updated the master branch here to use the same network code. I like it much better. It will connect on port 9090 by default. Many of the directions have changed, so be sure to update and re-read. Log data format and model have changed too.

tawnkramer, 2019-03-17:
Please let me know if new version not working for you.

closed 2019-03-17 by tawnkramer