tawnkramer/sdsandbox #19: Printing out error value to a file

JackWang1997, 2019-02-28:
Hi!

First of all sorry for not responding my last issue about NN not responding. I reinstalled everything and got it to work at last...

I am trying to compare my model's performance to the PID driving's performance. Then I realized that there are error values (prew err, abs total error) shown for pid driving, i think they mean how far is the car from the central line. Is this right?

I am thinking about to print out the prew err values for each frame for both PID driving and NN driving and then store those values to a file like txt file. I guess i then need to modify both PIDController file and also the file for driving NN. Is it possible to do this? If yes, how should i do this?

Thank you!

tawnkramer, 2019-02-28:
Hi Jack,

If you want to get the cross track errors, you might want to move to a different version of this project that gives those. Check out https://github.com/tawnkramer/donkey_gym and see if that gives you what you need. Check https://github.com/tawnkramer/donkey_gym/blob/b3e06b8c7c67d1a45b5da04325e6c2733f2768f1/donkey_gym/envs/donkey_sim.py#L183 and see that data['cte'] will have your cross track error.

Hope that helps,
Tawn

tawnkramer, 2019-03-01:
I've updated the master branch here, and you should be able to see data['cte] in predict server.py on_telemetry. Let me know if that solves your problem.

JackWang1997, 2019-03-07:
Thanks for replying! I went through the code you modified. It seems to me that now the images names' contain some datas like frame number, throttle, angle and mil. What does mil stands for? Also i think i may missed some part you changed since i still don't see where can i see cross track errors. What i am thinking is to compare the performance of the pid driving and the NN driving with datas like cross track error and angle.  So is there any way to store lists in a text file of these datas while running with pid/NN so that i can compare these datas easier? I checked your other version of this project. It is weird that i cant run it on my machine due to some software problem. It crashes  when i open it sometimes.

tawnkramer, 2019-03-17:
Jack I posted a direct link to the code where you can get CTE. Please reopen if I can help further. If you have a crash problem please report enough details so that I may reproduce.

closed 2019-03-17 by tawnkramer