tawnkramer/sdsandbox #21: unrecognized arguments: outputs/highway.h5

jewis123, 2019-04-15:
hello， how to fix this error? 
predict_server.py: error: unrecognized arguments: outputs/highway.h5

the error shows in this way
1. download this repo
2. install the requirements
3. cd sdsandbox
python src/predict_server.py outputs/highway.h5
4. the error shows
[image]

padalous, 2019-04-15:
I have the same error. Tried to look at the file, it's a "Hierarchical Data Format 5 File" (.h5), so tried to read it with pandas :

tawnkramer, 2019-04-15:
Add a —model= before the model name

jewis123, 2019-04-17:
wow excellent, it works!  Thx first, and what about the world complexity part ?Does it mean that I should replace the function of gen track to gen world?

tawnkramer, 2019-04-17:
Just a tiny bit more work than that...

jewis123, 2019-04-17:
OK ,I see. THX ;)

closed 2019-04-17 by jewis123