tawnkramer/sdsandbox #24: [question] Training loop on different random race tracks

tleyden, 2019-04-29:
How would I got about training a model using the approach from https://github.com/araffin/learning-to-drive-in-5-minutes in a training loop that trains the model on different tracks?

Here are the steps I'm currently using for training:

1. `train.py --algo sac -vae vae-level-0-dim-32.pkl -n 10000 --trained-agent logs/sac/DonkeyVae-v0-level-0_16/DonkeyVae-v0-level-0_best.pkl`
1. Run "Use NN Steering" via Unity UI
1. Wait until training finishes
1. Restart sdsandbox via Unity UI
1. `train.py --algo sac -vae vae-level-0-dim-32.pkl -n 10000 --trained-agent logs/sac/DonkeyVae-v0-level-0_17/DonkeyVae-v0-level-0_best.pkl`  (increment 16->17)
1. Repeat

Can this be done via a script that calls the Unity API?

tleyden, 2019-04-29:
I saw this comment https://github.com/tawnkramer/sdsandbox/issues/18#issuecomment-468330834 which seems like it might do what I was looking for.

tawnkramer, 2019-04-29:
Yes, definitely take a look at the donkey_gym wrapper. That may be useful to you, or it may just add a bunch of layers that get in your way. Let me know which.

Take a look at sdsandbox/src/predict_server.py for an example of sending messages to the simulator to control things. In particular send_regen_road will regenerate the random road with different curves and surfaces.

tleyden, 2019-05-18:
I've tried w/ no luck so far:

Here's my training loop attempt: https://github.com/tleyden/learning-to-drive-in-5-minutes/blob/c0b89213269b1fdec7077102d216e1eb14444cf5/train_loop.py#L186-L190

and here is the code I cherry picked from the copy of the donkey_gym simulator (I'm using https://github.com/araffin/learning-to-drive-in-5-minutes, which is using a copy of the donkey_gym wrapper.):

https://github.com/tleyden/learning-to-drive-in-5-minutes/commit/1c3ee94a700efda7409a43b7b8cd85f6e065b66b

I added some debug printlns to `TcpCarHandler.cs`, and I see these outputs in the unity console:

```
OnRegenRoad called with JSONObject
UnityEngine.Debug:Log(Object)
tk.TcpCarHandler:OnRegenRoad(JSONObject) (at Assets/Scripts/tcp/TcpCarHandler.cs:192)
tk.Dispatcher:Dipatch(String, JSONObject) (at Assets/Scripts/tcp/Dispatcher.cs:48)
tk.JsonTcpClient:Dispatch() (at Assets/Scripts/tcp/JsonTcpClient.cs:111)
tk.JsonTcpClient:OnDataRecv(Byte[]) (at Assets/Scripts/tcp/JsonTcpClient.cs:95)
tk.TcpClient:ReceiveCallback(IAsyncResult) (at Assets/Scripts/tcp/TcpClient.cs:71)
System.Net.Sockets.Worker:Receive()

RegenRoad called
UnityEngine.Debug:Log(Object)
tk.<RegenRoad>c__Iterator1:MoveNext() (at Assets/Scripts/tcp/TcpCarHandler.cs:209)
UnityEngine.MonoBehaviour:StartCoroutine(IEnumerator)
<Enqueue>c__AnonStorey1:<>m__0() (at Assets/Scripts/utility/UnityMainThreadDispatcher.cs:46)
UnityMainThreadDispatcher:Update() (at Assets/Scripts/utility/UnityMainThreadDispatcher.cs:34)

```

But they seem to be completely ignored and the road doesn't regenerate.   Is it because the neural net is still controlling the car, and that somehow blocks the road from regenerating?  Does it need to do the equivalent of hitting the Stop button before regenerating the road?

tleyden, 2019-05-19:
I found out why it wasn't working:

It was using random_seed = 0 in conjunction with this code:
 https://github.com/tawnkramer/sdsandbox/blob/1d2a2d77a57bb394796b0477ad52372425afba0a/sdsim/Assets/Scripts/tcp/TcpCarHandler.cs#L218

After commenting out that line, it works and the track is regenerated each time `donkeyEnv.regen_road()` is called.  Passing a non-zero random seed would probably also work too.

tawnkramer, 2019-05-19:
Sorry about that. I can see how that could be confusing. Can’t remember
offhand why zero is a special case.

tleyden, 2019-05-19:
Np.  Btw if you use `regen_road(rand_seed=int(time.time()))` instead of `regen_road()`, then it will also work as expected.  

0 isn't really a special case per-se, it's just that if you seed the random number generator with the same seed, then reinitialize it, it will end up with the same first pseudorandom number in the sequence over and over.

For example if with seed=99 the random number generator returned `[3.2, -5, 2, ..]` calling `InitState(99)` repeatedly and then getting the next generated number, you'll always get `3.2` back.  In the default case, it was doing the same thing except for `InitState(0)`.

tawnkramer, 2019-05-19:
Yes, that was the requested behavior by a researcher. They wanted to do
repeated runs that were as deterministic as possible. So you would need to
do your own seed calc and pass that in, or identical if you want to repeat
a previous run.

closed 2020-01-20 by tawnkramer