tawnkramer/sdsandbox #25: Several Missing References

tall-josh, 2019-05-05:
Unity Version: 2017.2.0b11 Personal (64bit)
OS: Ubuntu 18.04

Symptom:
Cannot start any of the driving environments. 
Steps:
1. Play From `Scene --> menu`.
2. Click the "Generated Track" option.

No screen change just stays on `menu` screen and the following errors.

Errors:
[image]

If I try the Warehouse track:
[image]

Upon startup, before running anything I get the following errors.
[image]

Any help would be greatly appreciated. I am currently looking to upgrade my Unity install.

Cheers Josh

tall-josh, 2019-05-05:
Also, upon opening `Scenes/avc2` and others I notice many missing prefabs. Is there a git-lfs pull or something I'm missing or something?

[image]

tall-josh, 2019-05-05:
Hmmmm, I just checked out the `dev` branch (was on the `donkey`) branch and everything seems to work

tleyden, 2019-05-19:
@tall-josh I think you want to be on the `donkey` branch.  The `dev` branch is 2 years old.  I'm using the `donkey` branch and have the basics working using Unity 2018.3.14f1.  (OSX 10.12.6)

tawnkramer, 2019-05-19:
It seems like his Unity version is too old. Please try a 2018 or newer.

The master branch was idle for a long time. I’ve tried to update it a few
weeks ago. Last I checked it was working.

But Mac has presented a special challenge since their awkward stance on
Nvidia drivers and cuda. I’ve mostly given up on it. Dual boot into Linux.

closed 2019-07-16 by tawnkramer