tawnkramer/sdsandbox #32: donkey gym dosen't work when building from source

Voyeger, 2019-09-13:
Hi, I am trying to build donkey branch of this simulator with unity 2018.3.4f1. Everything goes well except some '... is obsolete' warning from unity log console. The game runs fine itself except when I try it to use as a virtual donkeycar, the game starts and stays on the 'Menu' screen as if I start it manually (by double clicking on sim.x86_64). Nothing happens next..
The terminal window keeps showing "waiting for sim to start.." forever (though I can select a track from the game menu and play the game).
Please somebody help me :(

Here is my terminal output:

> $ python manage.py drive --model models/mypilot.h5
using donkey v3.1.0 ...
loading config file: /home/benzene/simulator/config.py
loading personal config over-rides

>config loaded
cfg.CAMERA_TYPE PICAM
cfg.CAMERA_TYPE PICAM
starting DonkeyGym env
donkey subprocess started
binding to ('0.0.0.0', 9090)
waiting for sim to start..
Preloaded type GtkListStore
Preloaded type GtkWindow
Preloaded type GtkVBox
Preloaded type GtkImage
Preloaded type GtkNotebook
Preloaded type GtkHBox
Preloaded type GtkFrame
Preloaded type GtkAlignment
Preloaded type GtkTreeView
Preloaded type GtkLabel
Preloaded type GtkCheckButton
Preloaded type GtkScrolledWindow
Preloaded type GtkComboBox
waiting for sim to start..
waiting for sim to start..
waiting for sim to start..
.....................................................

tawnkramer, 2019-09-13:
Instructions here for running:

http://docs.donkeycar.com/guide/simulator/

Voyeger, 2019-09-13:
@tawnkramer I already have followed the instructions and the downloaded binary from  'Donkey Gym Release' works fine.
I am only having problem when building it from source.

closed 2020-01-20 by tawnkramer