tawnkramer/sdsandbox #33: How to build a fixed road

wutianze, 2019-09-26:
Hi, I am using the sdsandbox to build my own car, it's amazing. However, I don't know how to draw a road instead of generating one. I find that in your code there is something that can generate roads from scripts but I don't know how to use it. Please help, thank you so much!

tawnkramer, 2019-09-26:
I’m happy you are enjoying it. It’s not too hard to make a track from a
script. The script goes in the resources dir. then you have to drag it into
the path manager in the scene. Check out the generated track scene for an
example. I can document a little more later.

wutianze, 2019-09-26:
So I should drag racePi for example to Path To Load in Path Manager and enable Do Load Script Path?

tawnkramer, 2019-09-26:
That’s sounds right.

wutianze, 2019-09-26:
Emmm,  it will come out with 
NullReferenceException: object reference not set to an instance of an object 
RoadBuilder.InitRoad:line 114
NullReferenceException: object reference not set to an instance of an object 
PathManager.GetPathEnd():line 113

closed 2020-07-29 by tawnkramer