tawnkramer/sdsandbox #39: Road not showing on linux

araffin, 2020-04-23:
Hello @tawnkramer ,

Building myself the binary or downloading the pre-built one, it seems that there is an issue with the road texture under linux:

[image]

That does not match what is seen in the doc...
However, not problem with the other levels...

araffin, 2020-04-23:
This issue is not present on windows:

[image]

tawnkramer, 2020-04-23:
The lack of track indicates there was an error while generating it. The
error was caused by reading a American encoded float into a system
expecting another floating point representation. You can see this by
checking the player log. And this is fixed with the latest sim. Check that
the version says 20.4.9

also, please use the Master branch now.

araffin, 2020-04-23:
> Check that the version says 20.4.9

How do you check that?
The home screen says "Pandemic edition v20.4"

I used the binary from https://github.com/tawnkramer/gym-donkeycar/releases/tag/v2020.v.9

tawnkramer, 2020-04-23:
that's the right url.
https://github.com/tawnkramer/gym-donkeycar/releases/download/v2020.v.9/DonkeySimLinux.zip

it should show 20.4.9 on menu screen.

araffin, 2020-04-23:
Ok, I re-build it from source (I had upgrade unity) and now it's working and I can see the full version, thanks =) (btw, it seems that the linux binary on github may be outdated...

closed 2020-04-23 by araffin

araffin, 2020-04-23:
I found out why I had the wrong binary: the [documentation](https://docs.donkeycar.com/guide/virtual_race_league/) does not point to the latest release but to the v20.4 (not 20.4.9)

tawnkramer, 2020-04-23:
Sorry about that. Thanks for the heads up. Will fix those docs.