tawnkramer/sdsandbox #40: How to add a new car model

naughtyStark, 2020-05-06:
Hi, first off, thanks for your hard work! the simulator is very useful in my research.

I'm new to unity as well as C#, and I was wondering what would be the steps for adding my own car model. For starters, I want to keep the underlying physics the same, just change the appearance(as in the shell, like have a Lamborghini shell instead of scion FRS), but once I get that done I would like to change the wheelbase as well. I understand that this may be a very simple question. I have looked up the general ways of including model files into the scene and so on but I would like to have the specifics as it would take a lot of manual tinkering to figure out which components I need to change/add and where in order to get what I want.

Thanks,
Regards,
Sidharth.

tawnkramer, 2020-05-06:
Well, to add the car to this repo, it needs to be free or something you
created and release to open source. You are welcome to add the car if you
keep it local, or release the binary.

This is not really a project for people who don't know Unity or C#. We can
make it easy to add a car shell in the future, but it's not there yet.

naughtyStark, 2020-05-06:
Hi,

The work will be kept local for a while. Could you give me any pointers on how I could replace the Scion FRS shell with anything else?( I already have the designs so getting the designs is not an issue).

Thanks,
Regards,
Sidharth.

naughtyStark, 2020-05-08:
Hi, I figured it out. It turned out to be easier than I thought. 
For anyone else who wants to do the same(Assuming you already have converted stl files to obj files):
1) Place your obj files in a new folder within the Assets folder (It just needs to be accessible when you open unity).  
2) Open the project in unity, Go to Assets-> Prefabs. Double click on "Donkey" 
3) Copy rc_CAR and paste under Donkey. Rename it to whatever name you want, for example, "new_car". Remove the objects contained within it. 
4) In the project tab (at the bottom) navigate to the folder where you have your designs
5) drag and drop them into the scene (which should still be the donkey scene from before).
6) rotate/translate/scale your objects. https://www.youtube.com/watch?v=uCA8Q3nwaz4 This video is particularly helpful in getting some understanding of how to deal with "GameObjects". In the hierarchy tab, drag the object under the "new_car" 
7) In the CarConfig.cs under Assets->Scripts, you can add an if-condition for your own body style and new variables for the car and its corresponding parts. The code, while in c#, is fairly simple to understand. 
8) Back in Unity, in the hierarchy tab, on the top left, if you click on Donkey, the inspector tab should show a bunch of stuff. Look for "Car Config (Script)". Here, you specify which variable in the C# script refers to which 3d object. If you've added your own variables, they will appear with blank fields. Click on the blank fields and a list should pop up. Find your 3d objects from that list. As an example, you can look at how the Rc_car_body and rc_car_parts fields are set. 

This should work as a basic guideline. Each case may be slightly different but a little bit of googling and tinkering should solve the problem :)

Thanks
Regards,
Sidharth.

closed 2020-05-08 by naughtyStark