tawnkramer/sdsandbox #45: Is there any good way to change the camera field of view?

hbzhang, 2020-07-22:
In our project, we want to change field of view, the question is that is there any good way to change the field of view of the camera that captures and log the images. If this is not possible would it be possible for us to set up multiple cameras to capture images.
Thanks!

tawnkramer, 2020-07-29:
there's an api to set the camera config.
https://docs.donkeycar.com/guide/simulator/#camera-config

closed 2020-07-29 by tawnkramer

AlexKaravaev, 2020-07-30:
@tawnkramer Could you please tell, what are the defaults parameters for the camera? I want to get the intristic and extrinstic params. 
I've found these in src/test_client.py
```
'{ "msg_type" : "cam_config", "fov" : "150", "fish_eye_x" : "1.0", "fish_eye_y" : "1.0", "img_w" : "255", "img_h" : "255", "img_d" : "1", "img_enc" : "JPG", "offset_x" : "0.0", "offset_y" : "3.0", "offset_z" : "0.0", "rot_x" : "90.0" }'
```
But I am not sure, that these are default ones.