tawnkramer/sdsandbox #49: Bad Json

araffin, 2020-08-24:
Hello,

I only experienced this issue when using an external server (ping > 100ms), for instance the one from https://www.twitch.tv/roboticists

from time to time (using the gym env), a bad json is received, for instance:
```
{"msg_type":"DQ","missed_cp":true{"msg_type":"telemetry",...}

```
Here there are two json messages inside one, causing the decoding to crash.
I'm not sure if this is due to packet loss or race condition when sending messages.

tawnkramer, 2020-08-24:
Do you get a print 
Exception: XXX
json: XXX
? It should do this and continue on when there's a json packet problem. It looks like your tcp packet stream somehow dropped a closing }. I'm not having luck reproducing it so far with the robotocists server.

araffin, 2020-08-24:
yes, that's how i could analyse it but that's not the issue.
it's also weird as we are using TCP and not UDP.
try connecting from Europe if you can

tawnkramer, 2020-08-25:
I was able to reproduce from aws in Paris w @160ms ping. I don't see anything wrong with the send on the server side, so I added code to recover the second good half of the packet. Not ideal, but it does recover some data there.

closed 2020-09-01 by tawnkramer